SICKAG/sick_safetyscanners #103: How to access Sick safety scanner data in docker container

harshkakashaniya, 2022-07-19:
I am able to do tcpdump and see the UDP packets in the container but I cannot access topic with this package. 

Host IP 192.168.0.5
sensor IP 192.168.0.3
container IP 172.20.0.5

Any solution?

puck-fzi, 2022-07-19:
Hi, this driver is not tested in a docker container.

With the information provided I cannot help you. I don't know how your actual setup is and how the ports and everything can communicate from your host to the docker container.

What does the console log show you when you start the ROS driver? Can it connect via TCP to the sensor or does it time out?

harshkakashaniya, 2022-07-19:
Thanks for quick reply...

I just went through this issue
https://github.com/SICKAG/sick_safetyscanners/issues/61

I was bridging port 6060:6060/udp
but it should be 192.168.0.5:6060:6060/udp

and while launching 

roslaunch sick_safetyscanners sick_safetyscanners.launch sensor_ip:=192.168.0.4 host_ip:=192.168.0.5 host_udp_port:=6060

I was providing docker IP instead of host IP.

Thank you very much though :)

closed 2022-08-22 by puck-fzi