SICKAG/sick_safetyscanners #104: Under what circumstances will this driver be killed by the system or to end running on its own?

COMING-X, 2022-08-01:
Hi
We used two lidars on AGVs. We created two nodes that distinguish the front and rear lidars.Both drivers can run normally and publish data when they are started. During running, only one node driver will end the operation many times one day, and the icmp host unreachable packets that are captured by wireshark about once a second, while the other node driver runs normally and reports text is normal.
Previously, the icmp host unreachable packet appeared through wireshark capture, and then we tried to reproduce this phenomenon to find the reason. We reproduced this problem by killing the driver, so we want to know what the driver has when it is running. Factors can cause it to be killed by the system or to end running on its own? 
Our CPU is a quad-core processor, and the load rate of one of the cores will reach more than 70%, because we have performed some CPU core binding operations on the processor. At first, the lidar driver did not bind any core to run. , so we tried to bind the lidar driver to a cpu core with a very low load rate to run, but the result was not what we expected, the problem will still appear, so we come to you for some solutions

puck-fzi, 2022-08-22:
Hi, sorry for the late response.

As I understand, you have a problem that the driver occasionally crashes. Which version are you using and can you provide a stack trace of the error? I would need more details of the actual error to provide you with information of what is happening. 

How are the drivers configured, are they using different ports? As of my understanding and tests, two drivers can run on the same machine.

Blues686, 2022-08-25:
The driver version is the latest we downloaded on August 1st. The stack trace is not available for you at the moment. We will try to do it. The ports are two different ports that are automatically selected according to the machine.  The configuration files of the two lidars are only different from sensor_ip and frame_id. The two lidars can indeed run on the same device, but the occasional crash of the driver only occurs on one of the nanoscans, while the use of the two microscans is very stable. has never had such a problem.The following is the configuration of the lidar.

sensor_ip = "192.168.30.10";
host_ip = "192.168.30.46";
host_udp_port = 0;
frame_id = "laser_front";
skip = 0;
angle_start = 0.0;
angle_end = 0.0;
time_offset = 0.0;
channel_enabled = true;
general_system_state = true;
derived_settings = true;
measurement_data = true;
intrusion_data = true;
application_io_data = true;
use_persistent_config = false;
channel = 0;
is_reverse_scan = true;
frequency_tolerance = 0.1;
expected_frequency = 20.0;

puck-fzi, 2022-08-25:
Thanks for the information.

Then it would be best to try to get the stack trace. It sounds like it might not be a problem of the two sensors and rather a problem, that the nanoscan and the driver do not work together as intended. With a stack trace I could further dive into where this happens and try to figure out why.

Your launch configuration look fine and should work as intended

puck-fzi, 2023-04-04:
Since there was no update with a stack trace I am closing this, please reopen if new information are agthered

closed 2023-04-04 by puck-fzi