SICKAG/sick_safetyscanners #107: Out of range bug still present in 1.0.8

reinzor, 2022-09-14:
Version: `1.0.8`

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

Follow up of https://github.com/SICKAG/sick_safetyscanners/issues/58

Bug still seems to be present in the latest version

I am also seeing these warning messages coming by:

```
[ WARN][/front_scan][1662478255.752680645]: Skipping data, sizes do not match, actual size is smaller then expected size! If this occurs please report with a stacktrace if the driver crashes at some other place. 
[ WARN][/front_scan][1662478255.754574851]: Expected minimum size: 5464
[ WARN][/front_scan][1662478255.755154463]: Actual size: 5060
[ WARN][/front_scan][1662478255.755987363]: Skipping all data for this message.

[ WARN][/front_scan][1662492345.148584654]: Skipping data, sizes do not match, actual size is smaller then expected size! If this occurs please report with a stacktrace if the driver crashes at some other place. 
[ WARN][/front_scan][1662492345.148653641]: Expected minimum size: 5108
[ WARN][/front_scan][1662492345.148678426]: Actual size: 5060
[ WARN][/front_scan][1662492345.148694592]: Skipping all data for this message.

[ WARN][/front_scan][1662520262.934361006]: Field Number Beams has a value larger then the expected Number of Beams for the laserscanners. Skipping this measurement.
[ WARN][/front_scan][1662520262.934413448]: Max expected beams: 2751
[ WARN][/front_scan][1662520262.934442805]: Number beams according to the datafield: -1056956303

[ WARN][/front_scan][1662674352.304261438]: Field Number Beams has a value larger then the expected Number of Beams for the laserscanners. Skipping this measurement.
[ WARN][/front_scan][1662674352.304358731]: Max expected beams: 2751
[ WARN][/front_scan][1662674352.304468138]: Number beams according to the datafield: -1899954145

[ WARN][/front_scan][1662765657.677277037]: Field Number Beams has a value larger then the expected Number of Beams for the laserscanners. Skipping this measurement.
[ WARN][/front_scan][1662765657.677343470]: Max expected beams: 2751
[ WARN][/front_scan][1662765657.677374916]: Number beams according to the datafield: 2031291393

[ WARN][/front_scan][1662796320.543376042]: Skipping data, sizes do not match, actual size is smaller then expected size! If this occurs please report with a stacktrace if the driver crashes at some other place. 
[ WARN][/front_scan][1662796320.543490062]: Expected minimum size: 5067
[ WARN][/front_scan][1662796320.543518815]: Actual size: 5060
[ WARN][/front_scan][1662796320.543539003]: Skipping all data for this message.
```

puck-fzi, 2022-09-14:
Hi,
I am still not sure where they originate from.
Could you provide me with the stack trace of where the program is crashing.
Otherwise I will not be able to implement a check to prevent the driver from crashing.

mechatheo, 2022-11-29:
I can confirm, we also keep seeing this error. 
[image]

we found out that we can trigger this error quite reliably if we call the `status_overview`  service in parallel. Increasing the service call frequency increased the crash likelihood for us.

FabianSchurig, 2022-12-05:
Still an issue. 🚀 

@puck-fzi are you able to reproduce it?

puck-fzi, 2024-06-19:
Hopefully, this is resolved through the PR https://github.com/SICKAG/sick_safetyscanners/pull/139

Please reopen if not.

closed 2024-06-19 by puck-fzi