SICKAG/sick_safetyscanners #110: ros driver consulting

liangchengcheng000, 2022-10-17:
Hi ,
we are using the nanoScan3 with the ros driver for Slam navigation, but it occurred below error during we launched the ros driver after power-on, below are the related error screen shots:
[image]
[image]

could you please help to check why?

thank you!

liangchengcheng000, 2022-10-17:
@Lennart Puck

puck-fzi, 2022-10-17:
Hi,

this sounds like you cannot reach the Sensor.
Are you able to ping the IP address of the sensor?

And is the IP address configured correctly in the launch file?

liangchengcheng000, 2022-10-17:
> Hi,
> 
> this sounds like you cannot reach the Sensor. Are you able to ping the IP address of the sensor?
### **_Yes,  we are sure the IP and the communication is ok, and this error happened sporadically, not every time(10 times/day during our commissioning). sometime we can launch the driver successfully._**

> And is the IP address configured correctly in the launch file?

**_we also checked the hardware connection, and all seems good.
could you please share more details about the error code we showed in the screen shot?
by the way, there is just one AGV with this nanoScan3 occurred this error, all other 20+ AGV are totally ok,  all configuration are the same._**

puck-fzi, 2022-10-17:
The error code states: Connection refused,

see here: https://www.ibm.com/docs/en/db2/11.1?topic=message-tcpip-errors

So there is mostly something interfering with the communication. Did you try connecting to the sensor with a different computer?

khlaifiabilel, 2023-02-21:
Greetings 

Unfortunately i am having the same problem as well, I am using the nanoScan3 Lidar sensor under a Ros Melodic distribution  for Slam tasks, but it am stuck with same error @liangchengcheng000 had. 

@puck-fzi, Can please help me identify the problem and how can i solve it, Thank you very much.

[image]

puck-fzi, 2023-06-01:
Where you able to get it working on your side?

puck-fzi, 2023-07-26:
Closing this due to inactivity

closed 2023-07-26 by puck-fzi