SICKAG/sick_safetyscanners #112: Ghost points seen in the laser scan

Soft-illusion, 2022-11-03:
Thank you for this package.
Ghost points are seen in laser scan with sick Microscan. But warning or safety is not triggered.
So how to access the filtered scan that sick safety is using on the ROS side (Not the raw scan)?

puck-fzi, 2022-11-03:
Hi, the ROS data is directly given by the sensor. So you will always get the raw data from the ROS driver.
However, triggering the warning and safety fields possibly needs more then one occurrence of a point in subsequent scans. But I am not sure on the internals of the SICK scanner to tell you the precise method which is used.

puck-fzi, 2022-11-03:
Hi, I don't know of any package directly. But maybe an extended kalman filter would be good here. There might be some implementations on github.

closed 2022-11-09 by puck-fzi