SICKAG/sick_safetyscanners #113: Costmap dynamic obstacle is not updated.

XieWup, 2022-12-16:
Hi, When we used sicknanoscan3 and pedestrians walking on the costmap, the obstacles in the cost map cannot be cleared in time, leaving a shadow track.
But if we use another radar(RPLIDAR S1), it OK
nacoscan3 topic:
[image]

RPLIDAR S1 topic:
[image]

Is it caused by angle_increment or other parameters?

puck-fzi, 2023-01-10:
Hi, this might be dependent on the node which creates your costmap. The topics of the two sensors will differ in their parameters. This might led to slightly different behavior in the creation. It might be just that the frequencies of both topics differ. Without further information it is hard to determine exactly what the problem is. However I don't think its a driver related problem. So maybe you can get better help at https://answers.ros.org/

XieWup, 2023-01-12:
Thank you for your reply. You are right. This is not caused by radar

closed 2023-01-12 by puck-fzi