SICKAG/sick_safetyscanners #115: out of rang access to scan points vector in `SickSafetyscannersRos::receivedUDPPacket`

kou-kikutake, 2023-01-26:
commit: d4afb8383d9265d62fbaed782f0b5079e2fbe55c

We observed `abort` is called in sick scanner node due to out of range access to `std::vector`.

Here is stack trace:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007fd0ea147859 in __GI_abort () at abort.c:79
#2  0x00007fd0ea3d0911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fd0ea3dc38c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fd0ea3dc3f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fd0ea3dc6a9 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fd0ea3d33ab in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000055f68ac61f9c in std::vector<sick::datastructure::ScanPoint, std::allocator<sick::datastructure::ScanPoint> >::_M_range_check (__n=<optimized out>, this=0x7fd0e5ba26a0)
    at /usr/include/c++/9/bits/stl_vector.h:1789
#8  std::vector<sick::datastructure::ScanPoint, std::allocator<sick::datastructure::ScanPoint> >::at (__n=<optimized out>, this=0x7fd0e5ba26a0) at /usr/include/c++/9/bits/stl_vector.h:1091
#9  sick::SickSafetyscannersRos::createLaserScanMessage (this=0x7ffe6123c3c0, data=...) at xxx/xxx/xxx/sick_safetyscanners/src/SickSafetyscannersRos.cpp:410
#10 0x000055f68ac685aa in sick::SickSafetyscannersRos::receivedUDPPacket (this=0x7ffe6123c3c0, data=...)
    at xxx/xxx/xxx/sick_safetyscanners/src/SickSafetyscannersRos.cpp:263
#11 0x00007fd0ea85793d in boost::function1<void, sick::datastructure::Data const&>::operator() (a0=..., this=0x55f68b243ae8) at /usr/include/boost/function/function_template.hpp:677
#12 sick::SickSafetyscanners::processUDPPacket (this=0x55f68b243ae0, buffer=...) at xxx/xxx/xxx/sick_safetyscanners/src/SickSafetyscanners.cpp:458
#13 0x00007fd0ea87e650 in boost::function1<void, sick::datastructure::PacketBuffer const&>::operator() (a0=..., this=0x55f68b248bc8) at /usr/include/boost/function/function_template.hpp:677
#14 sick::communication::AsyncUDPClient::handleReceive (this=0x55f68b2464b0, error=..., bytes_transferred=<optimized out>)
    at xxx/xxx/xxx/sick_safetyscanners/src/communication/AsyncUDPClient.cpp:116
#15 0x00007fd0ea87fbb9 in sick::communication::AsyncUDPClient::<lambda(boost::system::error_code, std::size_t)>::operator() (__closure=0x7fd0e5ba2c30, bytes_recvd=<optimized out>, ec=...)
    at xxx/xxx/xxx/sick_safetyscanners/src/communication/AsyncUDPClient.cpp:106
#16 boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>::operator() (
    this=0x7fd0e5ba2c30) at /usr/include/boost/asio/detail/bind_handler.hpp:164
#17 boost::asio::asio_handler_invoke<boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int> > (function=...) at /usr/include/boost/asio/handler_invoke_hook.hpp:69
#18 boost_asio_handler_invoke_helpers::invoke<boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>, sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)> > (context=..., function=...)
    at /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:37
#19 boost::asio::detail::asio_handler_invoke<boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>, sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int> (this_handler=0x7fd0e5ba2c30, 
    function=...) at /usr/include/boost/asio/detail/bind_handler.hpp:207
#20 boost_asio_handler_invoke_helpers::invoke<boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>, boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int> >
    (context=..., function=...) at /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:37
#21 boost::asio::detail::io_object_executor<boost::asio::executor>::dispatch<boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int>, std::allocator<void> > (a=..., f=..., this=<synthetic pointer>) at /usr/include/boost/asio/detail/io_object_executor.hpp:119
#22 boost::asio::detail::handler_work<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::asio::detail::io_object_executor<boost::asio::executor>, boost::asio::detail::io_object_executor<boost::asio::executor> >::complete<boost::asio::detail::binder2<sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::system::error_code, long unsigned int> > (handler=..., function=..., this=<synthetic pointer>) at /usr/include/boost/asio/detail/handler_work.hpp:72
#23 boost::asio::detail::reactive_socket_recvfrom_op<boost::asio::mutable_buffers_1, boost::asio::ip::basic_endpoint<boost::asio::ip::udp>, sick::communication::AsyncUDPClient::startReceive()::<lambda(boost::system::error_code, std::size_t)>, boost::asio::detail::io_object_executor<boost::asio::executor> >::do_complete(void *, boost::asio::detail::operation *, const boost::system::error_code &, std::size_t) (owner=0x55f68b244260, base=<optimized out>) at /usr/include/boost/asio/detail/reactive_socket_recvfrom_op.hpp:128
```

seems this code may access to `-1` -th element
```
  sensor_msgs::LaserScan    
  SickSafetyscannersRos::createLaserScanMessage(const sick::datastructure::Data& data)    
  {    
...
   
    scan.angle_min = sick::degToRad(data.getDerivedValuesPtr()->getStartAngle() + m_angle_offset);    
    double angle_max =    
      sick::degToRad(data.getMeasurementDataPtr()    
                       ->getScanPointsVector()    
                       .at(data.getMeasurementDataPtr()->getScanPointsVector().size() - 1)   
```

but not sure what condition causes such situiaton.

puck-fzi, 2023-01-26:
Hi, thanks for the stack trace. I will have a look at it. It might be related to: https://github.com/SICKAG/sick_safetyscanners/issues/107

So far, I was never able to reproduce the errors, so the stack trace does help me in at least catching the point where the error occurs.

mitchallain, 2023-08-30:
I can confirm that we are also seeing this and that @kou-kikutake's assessment is correct. This exception occurs when the scan points vector is empty. The console errors in #107 and #58 share the same conditions, as evidenced by
```
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

note the index value is the largest possible unsigned 64 bit value (size_t after wrapping).

We would still like to better understand the conditions that cause the scan points vector to be empty, but I can prepare a PR which checks the vector size before trying to access the last scan point to at least avoid crashing.

puck-fzi, 2024-06-19:
Hopefully this is resolved through https://github.com/SICKAG/sick_safetyscanners/pull/139

please reopen if not

closed 2024-06-19 by puck-fzi