SICKAG/sick_safetyscanners #117: Abnormal radar data publishing problems during operation

0n1whq, 2023-02-17:
The latest version 1.0.8 driver occasionally has abnormal radar data publishing problems during operation, which can be restored to normal after restarting the node, seriously affecting the operation. The specific analysis of the problem is as follows: 1. Check that the physical network connection between the host and the sensor is normal; 2. Use the Wireshark network analysis tool to monitor the data transmission between the host and the sensor. The packet length is normal and the transmission is normal; 3. Print the topic scan data of the node without information; 4. Restart the node to recover. The preliminary judgment is that it is a driving problem. Please check and improve it as soon as possible. Thanks!

Raymond601, 2023-02-21:
@Lennart Puck, could you please help to check the issue above? 
thank you!

Raymond601, 2023-02-21:
@puck-fzi
Lennart Puck

puck-fzi, 2023-02-21:
Hi, sorry for the late reply.
As you stated this is not normal behavior, I only experienced this once when my network connection was unstable and a disconnect happened.
Maybe this approach helps you: https://github.com/SICKAG/sick_safetyscanners/pull/109

Otherwise, I would need more Information about what is happening right before the abnormal behavior. Are there any network drops or other issues?

puck-fzi, 2023-02-21:
Hi, no the driver is not aware of sensor restarts or timeouts. This is nothing implemented. You would need to write a watchdog for this. It is currently not planned from my side

0n1whq, 2023-02-21:
> Hi, no the driver is not aware of sensor restarts or timeouts. This is nothing implemented. You would need to write a watchdog for this. It is currently not planned from my side

Thanks, Is it possible for the sensor to reset itself when the communication is normal?

0n1whq, 2023-02-21:
[image]
Hi,This picture shows that we encountered an exception when testing the data released by the radar node separately. The node can be restarted and returned to normal. The node has been running for about 35 hours. What is the reason for this?

puck-fzi, 2023-02-21:
> > Hi, no the driver is not aware of sensor restarts or timeouts. This is nothing implemented. You would need to write a watchdog for this. It is currently not planned from my side
> 
> Thanks, Is it possible for the sensor to reset itself when the communication is normal?

Hi, what the sensor internally does, i am not fully aware of. In my opinion though it should not reset itself

puck-fzi, 2023-02-21:
> [image] Hi,This picture shows that we encountered an exception when testing the data released by the radar node separately. The node can be restarted and returned to normal. The node has been running for about 35 hours. What is the reason for this?

This is probably related to #107.
if you have a stack trace for that issue, I would be happy to get that. Sicne i can not recreate the error on my system

0n1whq, 2023-02-22:
> This is probably related to #107. if you have a stack trace for that issue, I would be happy to get that. Sicne i can not recreate the error on my system

Hi, I have checked the relevant issues. It should be a kind of problem. I can locate the code segment where the problem occurs in SickSafetyscannersRos:: receivedUDPPacket by printing the log.

0n1whq, 2023-02-22:
> Hi, what the sensor internally does, i am not fully aware of. In my opinion though it should not reset itself

ok, Thanks.

puck-fzi, 2023-07-26:
Closing since it appears to be the same problem as issue #107

closed 2023-07-26 by puck-fzi