SICKAG/sick_safetyscanners #120: what's the transform of the scan link?

12cp38, 2023-03-08:
Hello, we have a nano scan 3 mounted on our robot. However,  when I move the robot, the laser scan (obstacles) rotates with the robot. There are two possibilities that could cause these obstacle rotations
1) odometry of the robot is incorrect. I have already verified that the odometry from the wheels encoders is okay because when I rotate the real robot 5 times, it 5 rotates 5 times in rviz(visualisation).
2) the transform (rpy) of the laser link with base_link is incorrect. I have placed the nano scan 3 facing upwards (not inverted) and I am setting rpy values of the joint to 0 and xyz to exact values w.r.t baselink. Now, when I rotate the robot, the results are given in the following pictures

[image]
[image]

My question is, is there any transform already present when we use ros driver for the scan frame? what should be the values of rpy in the joint between base_link and scan?

Thank you!

puck-fzi, 2023-03-13:
Hi, there is a frame for the laserscanner, which can be configured in the launch file.
However, for setting up the URDF and including the sensor at the correct place you would need to assure that this is setup correct by yourself, since this is robot specific.

farazxavor, 2023-03-15:
Yes but this frame is not according to ros convention (right-hand rule) x forward, y towards the left, and z upward. I want to know the axes of this frame. I mean which direction is x-axis and so on. can you comment on that?

puck-fzi, 2023-04-04:
I think it should support the ROS conventions. 
The sensor itself has a range from -47.5° to 227.5°, however the ROS driver supports the convention that 0 points straight forward. Therefore the measurements are adjusted to fir the range -137.5° to 137.5°.
And since we are in the ROS ecosystem, the measurements are given in radians

puck-fzi, 2023-06-01:
After testing once more, I can confirm that the x-axis points to the front and y-axis to the left.

It therefore is according to the ROS convention. Closing this, if there are any open points please reach out again

closed 2023-06-01 by puck-fzi