SICKAG/sick_safetyscanners #123: Probem reading data from MicroScan 3 using C++ API on Linux

geofftreen, 2023-03-21:
I have followed the directions and am able to connect and communicate with the sensor. I can send a findSensor API command, and I can see the MicroScan 3 blink multicolor lights in response. However, when I attempt to upload sensor settings (triggered by a SyncSickSafetyScanner object creation or calling changeSensorSettings directly), I get an error ("Command Method Not Accepted"). Here is my code:

    std::string sensor_ip_str = "192.168.1.186" ;
    sick::types::ip_address_t sensor_ip = boost::asio::ip::address_v4::from_string(sensor_ip_str);
    sick::types::port_t tcp_port {2122};

    // Prepare the CommSettings for Sensor streaming data
    sick::datastructure::CommSettings comm_settings;
    std::string host_ip_str = "192.168.1.100" ;
    comm_settings.host_ip = boost::asio::ip::address_v4::from_string(host_ip_str);
    comm_settings.host_udp_port = 0;
    comm_settings.e_interface_type = sick::datastructure::E_ETHERNET_IP;

    cout << "Creating a safety scanner" << endl ;
    // Create a sensor instance
    auto safety_scanner = std::make_unique<sick::SyncSickSafetyScanner>(sensor_ip, tcp_port, comm_settings);

Furthermore, calls to receive data packets time out with nothing received. In fact, the error I'm seeing seems similar to this unsolved issue raised on the Github repo: https://github.com/SICKAG/sick_safetyscanners/issues/23.

I am working in Linux using Ubuntu 20.04. I am able to run Safety Designer in Windows and see the scan data. 

Thanks in advance.

puck-fzi, 2023-04-04:
Hi, thanks for raising the issue, are you using the ros driver? or are you using the CPP driver (https://github.com/SICKAG/sick_safetyscanners_base) instead?

If you are using the ROS driver, could you try setting the parameter in the launch file? The return you are getting sounds like the settings you are passing to the sensor are rejected by your sensor itself. What is the sensor model you are using?

puck-fzi, 2023-07-31:
Closing, since its probably in the wrong repo. If this is still an issue, please reopen in the correct repo

closed 2023-07-31 by puck-fzi

marosko89, 2025-03-10:
I tried running the Sick_Safetyscanners_Base CPP Driver without ROS for the Synchronous Client, and I am only getting the response 
```
[INFO]: Command Method Acknowledged.
LiDAR driver initialized successfully.
Failed to fetch LiDAR data: Timeout exceeded while waiting for sensor data [timeout: 5seconds]
Failed to fetch LiDAR data: Bad file descriptor
Failed to fetch LiDAR data: Bad file descriptor
```
My `lidar.cpp` looks as follows:
```
/// @file lidar.cpp
/// @brief LiDAR scanner data sending.
///
/// Copyright © 2025, Asseco CEIT, a.s., all rights reserved.
/// See accompanying file LICENSE.txt for details.
///
#include "lidar.hpp"

LidarDriver::LidarDriver(const std::string& sensor_ip, uint16_t sensor_port, const std::string& host_ip)
    : _sensor_ip(sensor_ip), _sensor_port(sensor_port), _host_ip(host_ip) {}

// Join to LiDAR
bool LidarDriver::connect() {
    try {
        sick::types::ip_address_t sensor_ip = boost::asio::ip::address_v4::from_string(_sensor_ip);
        sick::types::ip_address_t host_ip = boost::asio::ip::address_v4::from_string(_host_ip);

        // Port configuration
        sick::datastructure::CommSettings comm_settings;
        comm_settings.host_ip = host_ip;
        comm_settings.host_udp_port = 0;

        // Initialization of the scanner
        _safety_scanner = std::make_unique<sick::SyncSickSafetyScanner>(sensor_ip, _sensor_port, comm_settings);

        std::cout << "LiDAR driver initialized successfully." << std::endl;
        return true;
    } catch (const std::exception& e) {
        std::cerr << "Failed to initialize LiDAR driver: " << e.what() << std::endl;
        return false;
    }
}

void LidarDriver::fetchData() {
    if (!_safety_scanner) {
        std::cerr << "LiDAR driver is not initialized." << std::endl;
        return;
    }

    try {
        // Timeoutu 5 sec
        auto timeout = boost::posix_time::seconds(5);

        // Read Data from Sick
        sick::datastructure::Data data = _safety_scanner->receive(timeout);

        std::cout << "Received LiDAR data:" << std::endl;
        // Processing

    } catch (const std::exception& e) {
        std::cerr << "Failed to fetch LiDAR data: " << e.what() << std::endl;
    }
}
```

and `main,cpp`

```
auto main() -> int {
    // Lidar test
    std::string sensor_ip = "192.168.0.11";
    uint16_t sensor_port = 2122;
    std::string host_ip = "192.168.0.9";     //my PC

    // Instation of LiDAR drivera
    LidarDriver lidar_driver(sensor_ip, sensor_port, host_ip);

    if (!lidar_driver.connect()) {
        std::cerr << "Failed to connect to LiDAR." << std::endl;
        return 1;
    }

    while (true) {
        lidar_driver.fetchData();
        std::this_thread::sleep_for(std::chrono::milliseconds(100));  // Waiting 100 ms between data loads
    }

    return 0;
}
```

Do you know what might be the problem?