SICKAG/sick_safetyscanners #129: Wrong min angle

zhibopotame, 2023-05-10:
Hello, 
in reconfigureCallback function in the ros node
`
if (config.angle_start == config.angle_end)
    {
      m_communication_settings.setStartAngle(sick::radToDeg(0));
      m_communication_settings.setEndAngle(sick::radToDeg(0));
    }
    else
    {
      m_communication_settings.setStartAngle(sick::radToDeg(config.angle_start) - m_angle_offset);
      m_communication_settings.setEndAngle(sick::radToDeg(config.angle_end) - m_angle_offset);
    }
`
the input start angle can be negative here, but the m_communication_settings.setStartAngle function's input is unint32_t, if we set start angle = -2.3998 and end angle slightly different, we will see a chunk of laser disapeared, please verify it there is a bug in the input value.

puck-fzi, 2023-06-01:
Hi,

i am not exactly understanding what the error is. When you only have a small window between your angles, the majority of the scan will disappear and only the area in between will be published. If the values are equal, the complete area will be published

zhibopotame, 2023-06-02:
@puck-fzi  Thanks for replying. What we've experienced is that by setting
`m_communication_settings.setStartAngle(sick::radToDeg(-2.3998) - m_angle_offset);
m_communication_settings.setEndAngle(sick::radToDeg(2.3998) - m_angle_offset);`
not setting them both to 0, we had a portion of laser disapeared as shown in the picture.
What I've been trying to say is, maybe to reproduce the same in ROS, just put the start angle and end angle slighly different in the dynamic reconfig, to avoid this condition `if (config.angle_start == config.angle_end)`, and see the laser coverage visualization in Rviz.
[image]

chenfumin196, 2023-06-07:
Hello,
I encountered the same problem, how do I set the values of angle_start and angle_end for an angle of 230 degrees in the launch file. thanks

puck-fzi, 2023-06-07:
> @puck-fzi Thanks for replying. What we've experienced is that by setting `m_communication_settings.setStartAngle(sick::radToDeg(-2.3998) - m_angle_offset); [...]

When I set the parameters as you suggest, i get the whole scan area. I will have a look however, if there is something, which is faulty.

puck-fzi, 2023-06-07:
> Hello, I encountered the same problem, how do I set the values of angle_start and angle_end for an angle of 230 degrees in the launch file. thanks

For a 230 degrees you should put the corresponding radian values in the launch file. So for your case your range would be [-2.0071285, 2.0071285]

chenfumin196, 2023-06-07:
Thank you  for replying. I have tried this value[-2.0071285, 2.0071285],this is the result
[image]
it seems that the angle_start values is unequal, compared to others, When I set a continuously changing value, the value of angle_end is displayed correctly ,this makes me feel confused.

Playfish, 2023-06-12:
Hi @puck-fzi , Is the following line correct?

https://github.com/SICKAG/sick_safetyscanners/blob/832d20f65ff76152561b8cb7ba7fe11a26164aa5/include/sick_safetyscanners/datastructure/CommSettings.h#L151

When I set start_angle with -0.677885881, the topic's angle_min is not -0.677885881.
Thanks.

Raymond601, 2023-06-20:
@puck-fzi would you please help to check above reply?
thank you very much!

puck-fzi, 2023-06-23:
I will check if this is according to the specifications of the sensor or if it should be a signed int. However, this will take another two weeks since I am currently on a conference. I'll try to get back to you as soon as possible

puck-fzi, 2023-07-10:
> Hi @puck-fzi , Is the following line correct?
> 
> https://github.com/SICKAG/sick_safetyscanners/blob/832d20f65ff76152561b8cb7ba7fe11a26164aa5/include/sick_safetyscanners/datastructure/CommSettings.h#L151
> 
> When I set start_angle with -0.677885881, the topic's angle_min is not -0.677885881. Thanks.

Hi, I am currently looking into this issue. Could you elaborate a bit more on what you are seeing.
I would need to know:
* Your input (as stated)
* Expected Result
* Experienced Result

The more details you can give me the better, is it a source build (if so, which commit) or is it installed. On which plattform, whic compiler version etc. The more details you can pass on the better i might be able to udnerstand whats going wrong

Playfish, 2023-07-21:
Hi @puck-fzi 
# Basic software information:
 * ROS version: 1.0.7
 * Lidar type: Sick safetyscanners nanoScan3 Pro

## Basic information before modification
The configuration of ros parameters in yaml:
 * angle_start: 0.0
 * angle_end: 0.0

The output `angle_min` and `angle_max` in ROS topic is:

 * angle_min: -0.8290314078330994
 * angle_max: 3.9706625938415527

## After modification:

When I change `angle_start` and `angle_end` to:
 * angle_start: -0.6290314078330994
 * angle_end: 3.5706625938415527

The output `angle_min` and `angle_max` in ROS topic is:
 
 * angle_min: -0.01454518735408783
 * angle_max: 3.566314697265625

# Expected Result:

The expected result above is that change `angle_start` should be equal with `angle_min` in ROS topic, like:

 * angle_start: -0.6290314078330994
 * angle_end: 3.5706625938415527

ROS topic's `angle_min` and `angle_max` should be:

 * angle_min: -0.6290314078330994
 * angle_max: 3.5706625938415527

Please tell me whether the above information is sufficient.
Thanks

puck-fzi, 2023-07-21:
Thank you for the information. Please verify that the results are the same under version 1.0.9. 

Either way, I will have a look into the error

Raymond601, 2023-07-24:
Hi @puck-fzi , we tried the newest version 1.0.9, but the result is the same...

puck-fzi, 2023-07-24:
Alright, thanks for letting me know

puck-fzi, 2023-07-24:
Please checkout the following branch and let me know if that fixed your issue: https://github.com/SICKAG/sick_safetyscanners/tree/fb_fix_for_angles

Playfish, 2023-07-25:
Thanks, for now I test about 35 times with 15 restart ROS node and 10 reconnect device with run ROS node.

A UDP connection occurred but no data was published at first time when I reconnect device and run ROS node.

Other times the topic has changed to correct `angle_start`(-0.629031) and `angle_stop`(3.770663)

Please let me know the UDP connection but not data was published whether it can be solved.

puck-fzi, 2023-07-26:
Thanks for letting me know, that the fix worked. 
I will integrate it into the main branch.

Once its merged, I will close this issue.
For the reconnects this is currently not implemented, however there is an untested MR https://github.com/SICKAG/sick_safetyscanners/pull/131 which includes a version of a reconnect

Playfish, 2023-07-28:
Hi, I found other issue is that nanoScan3 ros data of topic will rotate aboute 90°, like under images:
normal data:
[image]

rotate 90° with other nanoScan3:
[image]

This problem is extremely unlikely, can you help me find out why?
Thanks.

puck-fzi, 2023-07-31:
Hi, please open another issue for the new error.

puck-fzi, 2023-07-31:
Merged fix for min angle

closed 2023-07-31 by puck-fzi