SICKAG/sick_safetyscanners #133: ROS node crashed

renguorong, 2023-07-19:
ubuntu20.04，kernel：5.10，ros：noetic，sick_driver：1.0.8，node crashed

[image]
[image]

Raymond601, 2023-07-20:
@puck-fzi Could you please take a look at the issue above?
thank you！

puck-fzi, 2023-07-21:
Since you are using version 1.0.8. please update it to 1.0.9. this should fix the issue

It appears, that the .hpp is not correctly installed which should have been fixed in commit cfa456e78af42b82c1031354864837a798add4a7

puck-fzi, 2023-07-31:
Closing, since this should be fixed as stated above. Reopen if it persists