SICKAG/sick_safetyscanners #136: ros data rotate 90° compared with the actual contour

Raymond601, 2023-08-02:
Hi,
I found other issue is that nanoScan3 ros data of topic will rotate aboute 90°, like under images:
normal data:
[image]
rotate 90° with other nanoScan3:
[image]

This problem is extremely unlikely, can you help me find out why?
Thanks.

Raymond601, 2023-08-11:
@puck-fzi Could you please help to check above?
thank you!

puck-fzi, 2023-08-11:
Please refrain from tagging me in every issue. I will get a notification on every issue anyway.

To answer the issue please provide more information, and not just the result. Is it happening all the time with the same sensor, on every startup. Which sensor are you using? Which version of the driver? WHat are your launch files? 

There is a flag in a launch file which offsets the measurement by 90 degrees, to be according to the SICK convention. Is that set?

Playfish, 2023-08-15:
parameters of launch file:
```yaml
  <arg name="sensor_ip"             default="192.168.1.1" />
  <arg name="host_ip"               default="192.168.1.215" />
  <arg name="interface_ip"          default="0.0.0.0" />
  <arg name="host_udp_port"         default="0" />
  <arg name="frame_id"              default="scan" />
  <arg name="skip"                  default="0" />
  <arg name="angle_start"           default="0" />
  <arg name="angle_end"             default="0" />
  <arg name="time_offset"           default="0.0" />
  <arg name="min_intensities"       default="0.0" doc="minimal intensity for a laserscan point" />
  <arg name="channel"               default="0" />
  <arg name="channel_enabled"       default="True" />
  <arg name="general_system_state"  default="True" />
  <arg name="derived_settings"      default="True" />
  <arg name="measurement_data"      default="True" />
  <arg name="intrusion_data"        default="True" />
  <arg name="application_io_data"   default="True" />
  <arg name="use_persistent_config"   default="False" />
```

All sensors so far should have this problem.
Sensors: nanoScan3
Drivers version: 1.0.7

puck-fzi, 2023-08-23:
Hi, to figure out what the problem is, I still need more information.
The launch parameters look okay so far.

Do you experience the issue with each sensor you have? So it is not only one sensor but each sensor at random stages?
Is there anything special setup on the sensors which might influence the behavior? When do you notice the faulty behavior, at startup? at other times? Is it on a mobile system? Is it stationary?

Otherwise, I would advise you to update to the latest version 1.0.9 or, if you are using a source build the latest commit from this repository. However, there most likely is not a fix for your problem in there.

puck-fzi, 2024-06-19:
closing due to inactivity

closed 2024-06-19 by puck-fzi