SICKAG/sick_safetyscanners #137: Monitor case of Sick microscan3 ROS

nguyendo1403, 2023-08-24:
Hello, I am currently having difficulty sending service input to set the monitoring case when running the package sick_safetyscanners on ROS. I have configured on Safety designer software with 2 monitoring cases and cannot use it. . Can someone help me with an example program to set the supervisor instance? . Thank you

puck-fzi, 2023-08-24:
Hi,

As far as I know, the safety-related features can not be triggered via the Cola2 protocol. Therefore this can not be achieved with the ROS driver.
The possible Cola2 Methods can be found in 7.3 in the following document: https://cdn.sick.com/media/docs/1/01/701/Technical_information_microScan3_outdoorScan3_en_IM0083701.PDF

The monitoring cases and switches have to be defined inside the safety designer.

nguyendo1403, 2023-08-25:
@puck-fzi Can you share me a detailed example of changing the monitoring case when running on ROS? Thanks!

puck-fzi, 2023-08-25:
As stated above, the safety related features have to be set up in the safety designer. That includes the switches between monitoring cases. It is not possible to this inside the Ros driver or any other driver using the cola2 protocol

nguyendo1403, 2023-08-30:
@puck-fzi Is there any way to change the monitoring case by embedded computer? I use TCP/IP protocol to connect between embedded computer and SICKMicroscan3.

nguyendo1403, 2023-09-11:
@puck-fzi Is there any way to change the monitoring case by embedded computer? I use TCP/IP protocol to connect between embedded computer and SICKMicroscan3. Thank you!

puck-fzi, 2023-09-11:
Hi, as far as I know, that is not possible, but maybe ask your contact at SICK directly since I am unfamiliar with all the safety-related features and how to set them correctly. Since the driver is more focused on the UDP data parsing and output and the COLA2 protocol. I forwarded your issue as well, however, my contact is currently on vacation and will take some time to respond.

nguyendo1403, 2023-09-18:
Hi @puck-fzi !
I'm using Sick Microscan3 and ROS IPC. I want to use lidar for Lidar-loc and run package sick_safetyscanner.
But, when I run MS3 with Lidar-loc first. It's OK.
Then, I run package Sick_microscan3 =>Error: Lidar-loc can't get data from lidar.

My configuration:

MS3:
ip:192.168.3.4
ip address: 192.168.3.100
UDP port: 6060

ROS IPC:
ip:192.168.3.100
Please, refer to the attached pptx for more detailled description. Thank you very much!
[](url)
[LiDAR-LOC_ROS.pptx](https://github.com/SICKAG/sick_safetyscanners/files/12645238/LiDAR-LOC_ROS.pptx)

puck-fzi, 2024-06-19:
Closing due to inactivity

closed 2024-06-19 by puck-fzi