SICKAG/sick_safetyscanners #140: Error when using Microscan3 with Lidar-loc and running package sick_safetyscanner.

nguyendo1403, 2023-09-16:
Hi everyone!
I'm using Sick Microscan3 and ROS IPC. I want to use lidar for Lidar-loc and run package sick_safetyscanner. 
But, when I run MS3 with Lidar-loc first. It's OK.
Then, I run package Sick_microscan3 =>Error: Lidar-loc can't get data from lidar.

My configuration:
- MS3: 
      ip:192.168.3.4
      ip address: 192.168.3.100
      UDP port: 6060
      
 - ROS IPC:
     ip:192.168.3.100
Please, refer to the attached pptx for more detailled description. Thank you very much!
[](u
[LiDAR-LOC_ROS.pptx](https://github.com/SICKAG/sick_safetyscanners/files/12637012/LiDAR-LOC_ROS.pptx)
rl)

puck-fzi, 2023-09-19:
Hi, I am out of office until October.

Are loc and ros driver on the same device? Or are that different devices?.if they are different you could use a Multicast IP.

nguyendo1403, 2023-09-20:
Hi @puck-fzi, I run Lidar-loc and ros driver on the same device.

puck-fzi, 2023-10-04:
If they are on the same device this will only be possible if your sensor supports multiple channels( i.e. NanoScan3), otherwise the UDP port will be blocked by the ROS driver.

tbohnma, 2024-01-03:
There is a possible solution been provided in another thread by port forwarding. It has, however, not been tested
https://github.com/SICKAG/sick_lidar_localization/issues/8#issuecomment-1741190439

Feel free to provide feedback.

puck-fzi, 2024-06-19:
Closing due to inactivity, if there is feedback, please to the mentioned solution, please add it to the thread

closed 2024-06-19 by puck-fzi