SICKAG/sick_safetyscanners #141: Wrong received checksums in meta data

nicostudt, 2023-09-27:
Moin,

We currently have the issue that the checksums(both app_checksum and overall_checksum) do appear to use the opposite endianness for our Sick nanoScan3 I/O . As mentioned in #126 by @YannickdeHoop the checksums are received using big endian, but neither do I see that specified in the [technical specs](https://cdn.sick.com/media/docs/1/01/701/technical_information_microscan3_outdoorscan3_nanoscan3_data_output_via_udp_and_tcp_ip_en_im0083701.pdf) nor in the actual meta data of the scanners.

puck-fzi, 2023-10-04:
Thanks for opening this, I will check this, since I am sure, there was a reason why it was read the other way. I will have a look into it

chr-wurm, 2023-10-06:
Do you use the`config_metadata` service I added in https://github.com/SICKAG/sick_safetyscanners/pull/93/files? It swaps the bytes, as I did not fix the root cause in the parsing - I think we should rather change that back to non-swapping.

nicostudt, 2023-10-08:
> Do you use the`config_metadata` service I added in https://github.com/SICKAG/sick_safetyscanners/pull/93/files?

Yes, I do. The double swap is indeed unfortunate. ^^ 
Apart from that, I would like to know if the technical spec actually specifies the endianness like that. Maybe one of you could give me a hint with the passage in the document.

puck-fzi, 2024-06-20:
Hi, sorry for the late reply; just to clarify things now, what exactly is the issue now? Is there a double change of the order now? And the PR https://github.com/SICKAG/sick_safetyscanners/pull/142 is still necessary? Or are at different points in the Code appearing different outcomes due to multiple changes?

nicostudt, 2024-06-20:
Hey, no problem. 
I can only say that since we use the version from #142, we have no longer observed any issues with wrongly reported checksums.