SICKAG/sick_safetyscanners #143: Monitoring case switch using the ROS driver

pablo-sick, 2023-10-04:
Hi  Pablo Abia,

By the way, let me ask how to switch monitoring case of Sick microscan3
when i run ROS package sick_safetyscanners.
I have configured with 2 monitoring cases for sick microscan3 on Safety
Designer

Nguyen,

Vào Th 6, 29 thg 9, 2023 vào lúc 23:39 pablo-sick <
***@***.***> đã viết:

> Dear @nguyendo1403 <https://github.com/nguyendo1403> ,
>
> It's Pablo Abia, application engineer from LiDAR-LOC. I could reproduce
> the same issue on my PC. As you say, with your current configuration it
> does not seem possible to get the data on both LiDAR-LOC and the
> sick_safetyscanners ros driver...
>
> I also tried setting up the sensor to broadcast the output data (in Safety
> Designer, by configuring the target IP as xxx.xxx.xxx.255), but I came up
> with the same issue.
>
> One possible solution is to use a script to forward the traffic on
> 192.168.3.100:6060 to another port like 192.168.3.100:6061 and then
> configuring LiDAR-LOC to listen to one of them, and the sick_safetyscanners
> ros driver to listen to the other. On Github you can find some python
> scripts that performs this operation, like this one from the link below:
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/SICKAG/sick_lidar_localization/issues/8#issuecomment-1741190439>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/A7OG22XJRICT6PSX7Q3RSR3X432STANCNFSM5ACG5PEA>
> .
> You are receiving this because you were mentioned.Message ID:
> ***@***.***>
>

_Originally posted by @nguyendo1403 in https://github.com/SICKAG/sick_lidar_localization/issues/8#issuecomment-1741656443_

tbohnma, 2023-10-04:
Hello @nguyendo1403, 
It seems you raised this question in several issues. Please use the already provided answer in https://github.com/SICKAG/sick_safetyscanners/issues/137

Thanks!

puck-fzi, 2023-10-04:
Closing since its a duplicate, as pointed out.

closed 2023-10-04 by puck-fzi