SICKAG/sick_safetyscanners #150: SICK nano M1 error and lost data

Raymond601, 2024-09-24:
Hi ,

Raymond601, 2024-09-24:
[Uploading wireshark files.zip…]()

puck-fzi, 2024-09-25:
Dear Raymond,

which firmware of the nanoScan3 are you using? I guess it might be 1.92 or 1.93. There seems to be a bug on the sensor side with this firmware. If these are the firmware you are suing, please get in touch with your SICK contact, they might be able to help you. 

A similar problem has been seen in the ROS2 driver, see: https://github.com/SICKAG/sick_safetyscanners2/issues/48

Additionally, I can't see the images and wireshark data, they don't seem to be uploaded correctly, I simply see dead links.
Best regards

Raymond601, 2024-09-27:
Hi Puck-fzi,
Thank you for your information, I forward your recommendation to SICK AG, and I think they may get in touch with you these days.
one more quesion is that another engineer shared the ros driver document with me, but I cannot find the version inside it.
would you please help to check how can I find the ros revision？

Raymond601, 2024-09-27:
[Uploading SICK ROS.zip…]()

Raymond601, 2024-09-27:
[SICK ROS.zip](https://github.com/user-attachments/files/17159439/SICK.ROS.zip)

puck-fzi, 2024-09-27:
Dear Raymond, I don't quiet understand the ROS version you mean, If you have the source code, you can look in the CHANGELOG.rst in the main folder, there it should state the latest version. According to your files its the version 1.0.8 for the ROS1 driver.

Could you try to resend the images from the initial issue, since I can't open them

Raymond601, 2024-12-12:
Hi Puck-fzi,
Sorry for lately reply, I've uploaded the images and wireshark files in initial chatbox.
at the same time